Xilinx/Vitis-Tutorials #362: Different LUT utilization in the System Diagram and Utilization tabs

m-kru, 2023-02-14:
I have compiled and linked some custom kernel. When I view the report I can see different utilization information in the `System Diagram` tab and in the `Utilization` tab.

[image]

[image]

The first one presents that the LUT utilization is 4.85%.
However, the second one presents it is 16.25%. 
Other resources also have different values in both tabs, however the differences are not so big.

Is it a bug, or am I missing something?

```
- OS: 5.15.0-60-generic #66~20.04.1-Ubuntu
- Software Version: 2021.2]
- Alveo U50
```

imrickysu, 2023-02-20:
Hi @m-kru , on your second screenshot, the post route utilization is the device utilization. It includes kernels and the platform resources.

m-kru, 2023-02-21:
So the kernel resource utilization is less than 5%, but approximately 11% is taken by extra things such as AXI infrastrucute etc.?

AnusheelXilinx, 2024-12-21:
Hi @m-kru, as the query above is not related to the Vitis-Tutorial design, I request you to post any open queries in Vitis Community Forums:
https://adaptivesupport.amd.com/s/topic/0TO2E000000YKXhWAO/vitis?language=en_US

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx