Xilinx/Vitis-Tutorials #363: How to generate a .XO RTL kernel using verilog codes together and connects to a HLS kernel

tangyuelm, 2023-02-17:
Dear Authors,

Thank you for the tutorial, we now have a project on the U280 FPGA board involving a kernel writing in HLS and a kernel writing in Verilog. I am not sure how to build the project with both a HLS kernel and a Verilog kernel (The who kernels are connected via simple wires) with the current Vitis tool flow.
I found a tutorial in https://github.com/Xilinx/Vitis-Tutorials/tree/2021.1/Hardware_Acceleration/Feature_Tutorials/02-mixing-c-rtl-kernels showing how to combine a HLS kernel with a generated RTL kernel. However, it seems that the RTL kernel is also generated from HLS codes. I also referred to the tutorial from https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow/package_ip.md but cannot find any instructions about how to directly generate a .XO file from a hierarchy of Verilog modules. 
I also find a tutorial in https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Hardware_Acceleration/Design_Tutorials/03-rtl_stream_kernel_integration/doc/rtc_gen_tutorial.md but still cannot solve my problem. We are now using Vitis 2021.2 on the ubuntu system and it seems that the Vivado IDE of this version does not have U280 board, while the Vitis IDE only generates RTL from HLS. I also tried lower Vitis versions and the Vivado IDE of Vitis below 2020.2 does not have the operation of 'Generate RTL Kernel' from the Flow Menu
Could anyone provide any help? Thank you,

imrickysu, 2023-02-20:
Hi @tangyuelm , https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow/package_ip.md looks like a good reference for your use case. Could you clarify the need of "how to directly generate a .XO file from a hierarchy of Verilog modules"? If you add multiple files to the Vivado project, as long as these files form a module with hierarchy, this flow works.

Hi @Rampagee , do you have any further comments?

Rampagee, 2023-02-20:
Hi, @tangyuelm , do you mean you want to connect a RTL design and a HLS based design, then wrap these two modules into a unified design, then package it into a unified XO kernel? So I think you could instantiate you own RTL module and the synthesized RTL design from HLS in a RTL wrapper manually, then go through the Vivado RTL kernel packaging flow to get Vivado IP and Vitis kernel (XO). You could refer to https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/RTL-Kernel-Development-Flow or https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/doc/krnl_cbc.md for GUI based and Tcl based IP/kernel packing flow. 

Thanks

Gang.

Rampagee, 2023-02-20:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

[AMD Official Use Only - General]

Comments added in github.

Thanks,

Gang

tangyuelm, 2023-02-20:
> Hi @tangyuelm , https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow/package_ip.md looks like a good reference for your use case. [...]

Thank you for the reply. I have some Verilog codes that need to be packed together with an HLS code. I start with pack the Verilog codes (.sv files) into a .XO file. I am currently follow the link you mentioned and use the Vivado IDE to package. The Vivado 2021.2 does not have the Alveo boards so I have to try the Vivado 2020.2. However, in the "Enable Package for Vitis", this is no such an option of Package for IPI and Ignore Freq_Hz. 
Could you provide any suggestions?
[image]

Rampagee, 2023-02-20:
For Vivado 2021.2 release, please go to https://www.xilinx.com/member/alveo-vivado.html website to download Vivado board files, extract them, and define the board files directory in Vivado GUI. Thanks.

tangyuelm, 2023-03-03:
It works now. Thank you.

closed 2023-03-03 by tangyuelm