Xilinx/Vitis-Tutorials #364: failed software emulation in vitis : failed to find System Metadata section in   binary_container_1.xclbin

yangyazhou97, 2023-02-18:
The "Emulation-SW" build operation failed and the console said it couldn't find "binary_container.xclbin" file, but I do can find this xclbin file in the path, how can I solve this problem?

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Your environment: **
- OS:  Ubuntu 18.04
- Software Version:  2021.2
- Hardware : ZCU102

imrickysu, 2023-02-20:
Hi @yangyazhou97 , are you trying to reproduce any tutorials in this repo? If so, please share the design name or path. Thanks.

yangyazhou97, 2023-02-20:
Actually, I'm reproducing project form here: https://github.com/ATaylorCEngFIET/Building-Accelerated-Applications-with-Vitis/blob/master/Building%20Accelerated%20Applications%20with%20Vitis%20Lab.pdf

imrickysu, 2023-02-20:
For this generic question, we recommend you ask in forums.xilinx.com. This issue board focuses on the issues introduced by the tutorials in this repository.

Generally, if you use default build settings from the Vitis IDE, the generated xclbin can be emulated. XCLBIN is a container with multiple sections. From the error message, it looks like the xclbin you generated doesn't contain this section. You can check with the lab provider or forums.xilinx.com to help you investigate. I'm closing this issue.

closed 2023-02-20 by imrickysu