Xilinx/Vitis-Tutorials #365: failed to launch hardware emulator in vitis : failed to transfer files from host to target linux agent

yangyazhou97, 2023-02-20:
I want to launch hardware emulator in vitis, but it failed as shown in the picture, how can I solve this problem? 

[image]

imrickysu, 2023-02-20:
Hi @yangyazhou97 , the emulation in Vitis IDE takes several steps. It calls the emulation script to launch emulator for PS, PL (and AI Engine), using socket to connect and sync between each part. Vitis IDE needs to connect to the TCF Agent in Linux in QEMU, which emulates PS. From your error message, it looks like the Linux doesn't boot properly. You need to check the QEMU console for more details. 

Vitis-Tutorials -> Getting Started -> Vitis has steps related to hardware emulation.
https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/Getting_Started/Vitis/Part4.md
If you encounter issues when running tutorials in this repo, please let us know. For generic questions, please visit https://support.xilinx.com/s/ for further support.

closed 2023-02-20 by imrickysu

nuaaceieyty, 2025-01-30:
I got the same error