Xilinx/Vitis-Tutorials #366: 10-aie-dsp-lib-model-composer: Facing compilation errors while generating hardware image for the 10-aie-dsp-lib-model-composer tutorial

Hari-MathWorks, 2023-02-20:
** Which tutorial are you running? **
I'm running **Designing with the AI Engine DSPLib and Vitis Model Composer**(_https://github.com/Xilinx/Vitis-Tutorials/tree/c2e1a5d0caff6a2ec0bb196d0d807d733f9d0c3d/AI_Engine_Development/Feature_Tutorials/10-aie-dsp-lib-model-composer_) tutorial. 

** Describe the issue **
I was successfully ran the steps mentioned in the "Designing with the AI Engine DSPLib and Vitis Model Composer" tutorial. This tutorial shows the Codegen AIE Simulation in step 3. As a next step, I have tried generating hardware image(It was not described in the tutorial). But, I was facing the compilation errors while generating the hardware image.

**Please see the below text file for more details:
[hwImageGenerationLog.txt](https://github.com/Xilinx/Vitis-Tutorials/files/10782829/hwImageGenerationLog.txt)**

** To Reproduce **
1. Install "Model Composer v2022.1"
2. Install "MATLAB R2021b"
3. Create the model by following the steps mentioned in the **10-aie-dsp-lib-model-composer** tutorial.
4. Open "Vitis Model Composer Hub" block in the Simulink model and configure the model as follow:

- [image]
- [image]
- [image]
- [image]

** Your environment: **
- OS: Debian 11
- Software Version: Model Composer v2022.1
- Hardware (if applicable): VCK190

Thanks,
Hari

imrickysu, 2023-02-20:
Hi @OTremois , could you help with this question?

In the error log, it shows the following:

```
INFO: [v++ 60-2256] Packaging for hardware
ERROR: [v++ 60-2254] File specified by package.ps_elf option is not valid: main.elf
```

Hari-MathWorks, 2023-03-13:
Hi @imrickysu , @OTremois ,

Any update on this?

OTremois, 2023-03-13:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

[AMD Official Use Only - General]

Hi Hari,

This design was intended for 2022.2 tools suite. Apparently it fails with the 2022.1. May be one of the block is not compliant with the hardware flow (which was in its early days at that time) in 22.2.

Olivier

AnusheelXilinx, 2024-12-21:
Closing as there are no further queries.

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx