Xilinx/Vitis-Tutorials #369: AI Engine GMIO Performance Profile: Measurements using timer do not provide the listed performance

satoru-i, 2023-02-28:
** Which tutorial are you running? **
https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/02-using-gmio/perf_profile_aie_gmio.md
using macro __TIMER__.

** Describe the issue **
Measurements using timer do not provide the listed performance.
I measured it twice, and both times the performance was lower than stated.
Performance is particularly low when num=1 for the first time.

[image]
[image]

** To Reproduce **
export ROOTFS=xilinx-versal-common-v2022.1/rootfs.ext4
export IMAGE=xilinx-versal-common-v2022.1/Image
. xilinx-versal-common-v2022.1/environment-setup-cortexa72-cortexa53-xilinx-linux
make package TARGET=hw
cd sw
gedit Makefile &
`CXXFLAGS += -std=c++14 -D__TIMER__ -I$(XILINX_VIVADO)/include/ -I${SDKTARGETSYSROOT}/usr/include/xrt/ -O0 -g -Wall -c -fmessage-length=0 --sysroot=${SDKTARGETSYSROOT} -I${XILINX_VITIS}/aietools/include ${HOST_INC}`
make clean
make
cd ../
cp host.exe host_timer.exe 
copy host_timer.exe to the SD card.
booting Linux from an SD card
export XILINX_XRT=/usr
cd /run/media/mmcblk0p1
./host_timer.exe a.xclbin

** Your environment: **
- OS: Red Hat Enterprise Linux Workstation release 7.5
- Software Version: 2022.1
- Hardware (if applicable): VCP190 production board

AnusheelXilinx, 2024-12-21:
Hi @AlphaBu, may I know if the recent push for 2024.2 still has this issue?

Thanks
Anusheel

AlphaBu, 2025-03-27:
The test case is updated in later releases. And the performance number is updated too. The number may vary every run.