Xilinx/Vitis-Tutorials #370: Burst Understanding (Generic Question)

MahmoudElShimy96, 2023-03-02:
I saw in file Vadd_A_B.v in tutorial [01-rtl_kernel_workflow] that burst is recommended to be incremental only, why incremental is recommended and Can i use fixed burst ?

randyh62, 2023-03-02:
You can try it if you like. The recommendation suggests why the default is the best way to go. In addition, changing the burst may require you to change the associated testbench, and possibly the m_axi adaptor. 

Here is some [additional information](https://community.arm.com/support-forums/f/architectures-and-processors-forum/4594/difference-between-fixed-and-incr-burst-in-axi) you may find helpful. Here the access is with a "normal memory device."

AnusheelXilinx, 2024-12-21:
Closing as there are no further queries. 

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx