Xilinx/Vitis-Tutorials #373: Checking timing violation path

m-kru, 2023-03-02:
I have received negative slack for my kernel. Is there any way to identify logic path causing the timing violation from the Vitis HLS GUI? I feel like I have gone though all reports, clicked all possible icons, but still no info on operations causing the timing violation.

imrickysu, 2023-03-31:
Hi @randyh62 , could you help with this question?

m-kru, 2023-03-31:
@imrickysu The one that Xilinx wipes out once every few years, so that people cannot see how many bugs are in the Xilinx software?

randyh62, 2023-03-31:
Which version are you working with?

m-kru, 2023-04-03:
2021.2

randyh62, 2023-04-03:
You can run Vivado Synthesis and Implementation to generate timing reports from those tools. You can do this from the Implementation menu in the Flow Navigator, or by using the config_export command and then running export_design. I am attaching some screenshots from the 2021.2 release for your review. 
[image]
[image]

AnusheelXilinx, 2024-12-21:
Closing as there are no further queries. 

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx