Xilinx/Vitis-Tutorials #380: Vitis-Tutorials/Getting_Started/Vitis_HLS/synth_and_analysis Pre-synthesis Control flow does not show up

Eggwane, 2023-04-17:
Hi, 
Following the tutorial, [Vitis-Tutorials](https://github.com/Xilinx/Vitis-Tutorials)/[Getting_Started](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/Getting_Started)/[Vitis_HLS](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/Getting_Started/Vitis_HLS)/synth_and_analysis.md

Everything good but the pre-symthesis Control flow does not show up. I'm using Ubuntu 22.04.2 LTS. There's no error so it seems it's a different issue than #121 .

randyh62, 2023-04-17:
So you enable it, but nothing shows up. No error, no messages? Anything in the log file?

Eggwane, 2023-04-24:
Yes, I have nothing appearing, no error or anything in the log files.
The tab "Presynthesis Control Flow" shows up, I have everything (the left part with the tree (dct 2d/1d, read/write_data), the buttons on the top) except the flow chart : instead this part remains empty. 
I can provide you more information if you tell me what.

randyh62, 2023-04-24:
System details like OS and software version would be helpful.

closed 2023-05-11 by randyh62