Xilinx/Vitis-Tutorials #387: AIE A-to-Z 05 RPU BD Tcl edits fail with 2022.1

marctylerb, 2023-05-30:
** Which tutorial are you running? **
https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/01-aie_a_to_z/05-RPU%20version%5Boptional%5D.md

** Describe the issue **
Below Block Design script to edit Block Design to use RPU fails to complete. There are multiple lines with objects that do not exist possibly the script is referencing an older version.

** To Reproduce **
Open custom platform project generated with "01" step in tutorial creating a custom VCK190 platform. 

https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/01-aie_a_to_z/01-custom_base_platform_creation.md

After this, open Block Design and run script commands.

`set_property CONFIG.REMAPS {{M00_AXI {{0x4000_0000 0x200_0000_0000 1G}}}} [get_bd_intf_pins /ps_noc/S01_AXI]
set_property range 1G [get_bd_addr_segs {versal_cips_0/DATA_RPU0/SEG_ps_noc_C0_DDR_LOW0}]
assign_bd_address -offset 0x40000000 -range 0x40000000 -target_address_space [get_bd_addr_spaces versal_cips_0/DATA_RPU0] [get_bd_addr_segs ai_engine_0/S00_AXI/AIE_ARRAY_0] -force
assign_bd_address
validate_bd_design`

Error

`# WARNING: [BD 5-232] No interface pins matched 'get_bd_intf_pins /ps_noc/S01_AXI'
# # set_property CONFIG.REMAPS {{M00_AXI {{0x4000_0000 0x200_0000_0000 1G}}}} [get_bd_intf_pins /ps_noc/S01_AXI]
# ERROR: [Common 17-55] 'set_property' expects at least one object.
# Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.`

More errors are seen if the first line is fixed by changing "/ps_noc/S01_AXI" to "/**ci**ps_noc/S01_AXI".

OS: Ubuntu 18.04.1 LTS
Software Version: 2022.1
Hardware : VCK190

imrickysu, 2023-06-15:
Hi @xflorentw , could you check the command `set_property CONFIG.REMAPS {{M00_AXI {{0x4000_0000 0x200_0000_0000 1G}}}} [get_bd_intf_pins /ps_noc/S01_AXI]` and make sure the pin name matches Vivado example? The Vivado example design may be slightly different version to version.

marctylerb, 2023-06-15:
I think it's just a matter of different naming in the NoC instance and AXI paths. The below seems to work:
`set_property CONFIG.REMAPS {} [get_bd_intf_pins /cips_noc/S06_AXI]
set_property CONFIG.REMAPS {{M00_AXI {{0x4000_0000 0x200_0000_0000 1G}}}} [get_bd_intf_pins /cips_noc/S06_AXI]
set_property range 1G [get_bd_addr_segs {/CIPS_0/LPD_AXI_NOC_0/SEG_noc_ddr4_C0_DDR_LOW0}]
assign_bd_address -offset 0x40000000 -range 0x40000000 -target_address_space [get_bd_addr_spaces /CIPS_0/LPD_AXI_NOC_0] [get_bd_addr_segs ai_engine_0/S00_AXI/AIE_ARRAY_0] -force
set_property -dict [list CONFIG.CONNECTIONS {M00_AXI { read_bw {1720} write_bw {1720} read_avg_burst {4} write_avg_burst {4}} M00_INI { read_bw {128} write_bw {128}} }] [get_bd_intf_pins /cips_noc/S06_AXI]
assign_bd_address
validate_bd_design
`
[image]

AnusheelXilinx, 2025-01-10:
Tutorial was updated to fix the issue. 

Thanks
Anusheel