Xilinx/Vitis-Tutorials #388: Some questions about step 3 of "The Source Code for the Host Program"

Heng-Zhou, 2023-05-31:
I have some questions about step 3 of section "[The Source Code for the Host Program](https://github.com/Xilinx/Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/Part3.md#the-source-code-for-the-host-program)" of Part 3 of Vitis Getting Started Tutorial.

1. "These operations are enqueued in the command queue declared in Step 1"
I did not see any "command queue" mentioned in Step 1. What are they?

2. "The commands are put in the queue ..."
What is the statement in the source code that puts these commands in the queue?

3. "Because the queue used in the host code in this example is an ordered queue"
Why is this not-shown queue "an ordered queue"? Which statement sets that?

4. "The call to `q.finish()` ..."
Where is the call in the source code host.cpp to "`q.finish()`"?

Thanks and looking forward to clarifications.

randyh62, 2023-05-31:
Hi, That text is not correct as it applies to a prior OpenCL based code structure. The host code was updated to use the XRT native API and does not support the command queue. You can refer to the documentation at this location: https://xilinx.github.io/XRT/master/html/xrt_native_apis.html

There is a discussion of a queue feature that can be found at this location (but it is not featured in this tutorial): https://xilinx.github.io/XRT/master/html/xrt_native_apis.html#asynchornous-programming-with-xrt-experimental

Heng-Zhou, 2023-05-31:
Thanks for the reply. I hope the tutorials text can be updated to match the workings of Vitis 2023.1. Thanks again and have a good night.

closed 2023-05-31 by Heng-Zhou