Xilinx/Vitis-Tutorials #389: A  question about std::memcmp function call in the source code for the host program of Vitis getting started tutorial

Heng-Zhou, 2023-05-31:
The source code [host.cpp](https://github.com/Xilinx/Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/example/src/host.cpp) for the host program of Vitis getting started tutorial calls `std::memcmp` on line 74:

`if (std::memcmp(bo2_map, bufReference, DATA_SIZE))`

Here the third argument is `DATA_SIZE` which is the number of data items. But the C++ standard specifies that this argument should be the number of bytes to be compared ([here](https://en.cppreference.com/w/cpp/string/byte/memcmp)):

[image]

If we conform to the C++ standard, the third argument should be variable `vector_size_bytes` which is the number of bytes of the memory. So I was wondering whether it is a bug, or Xilinx redefined the meaning of the third parameter of `std::memcmp` function.

randyh62, 2023-05-31:
Hi, Have you run it with vector_size_bytes?

Heng-Zhou, 2023-06-04:
> 

I run it with `vector_size_bytes` under software emulation and hardware emulation, and it returns successfully.
It crashes on real hardware target at line 30, but that's not related to my question. I will ask this in another ticket.

imrickysu, 2023-06-15:
Hi @randyh62 , I think @Heng-Zhou is correct. 

`DATA_SIZE` is per int.
`vector_size_bytes` is per byte.

When doing `memcmp`, we can use `vector_size_bytes` or `sizeof(bufReference)`.

imrickysu, 2023-06-20:
fixed by 22a3dcc

closed 2023-06-20 by imrickysu