Xilinx/Vitis-Tutorials #400: Please remove duplicated contents

Heng-Zhou, 2023-07-12:
I found duplicated contents in webpage [Vitis Flow 101 – Part 1: Essential Concepts](https://github.com/Xilinx/Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/Part1.md#vitis-flow-101--part-1-essential-concepts).

First, in section "[Understanding the Vitis Programming and Execution Model](https://github.com/Xilinx/Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/Part1.md#understanding-the-vitis-programming-and-execution-model)", sentence "On data center platforms, the software program runs on an x86 server, and the kernels run in the FPGA on a PCIe®-attached acceleration card." were repeated twice.

Second, in section "[Understanding the Vitis Build Process](https://github.com/Xilinx/Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/Part1.md#understanding-the-vitis-build-process)", the paragraph starting with "The FPGA binary is built using the Vitis compiler." and the following figure are repeated twice.

So, if there is no special purpose for these duplication, can I ask the people who maintains the website to remove the duplicated contents? Thanks.

randyh62, 2023-08-02:
I have corrected the issue. Thanks for letting us know.

allyzhou, 2023-08-03:
This issue has been fixed by this commit https://github.com/Xilinx/Vitis-Tutorials/commit/07eaf0bc05fd882965a7306c0ac3ddaf1f0e62e7 
Thanks for letting us know.

closed 2023-08-03 by allyzhou