Xilinx/Vitis-Tutorials #403: Is Vitis 22.2 backward compatible with old OpenCL based host code?

dj-park, 2023-07-20:
I recently installed Vitis 22.2 but want to use old OpenCL based host code like (https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Getting_Started/Vitis/example/src/host.cpp).

I can confirm that the host code with XRT native API (https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/Getting_Started/Vitis/example/src/host.cpp) works properly with Vitis 22.2. But when I use OpenCL based host code, it doesn't properly returns platforms, meaning that `platform.sizes()` is zero, in [this loop](https://github.com/Xilinx/Vitis-Tutorials/blob/c84a09e50fdbbf7d1752740bea39218b2b639788/Getting_Started/Vitis/example/src/host.cpp#L139C1-L140C1). 

I can confirm that in Vitis 22.1 version, I didn't have this issue. Is there a quick fix for this?

randyh62, 2023-07-21:
Have you looked at the 22.2 XRT OpenCL API to see if anything changed between 22.1 and 22.2? 
https://xilinx.github.io/XRT/2022.2/html/index.html
Or maybe look at some of the 2022.2 Vitis_Accel_Examples:
https://github.com/Xilinx/Vitis_Accel_Examples/blob/2022.2/cpp_kernels/gmem_2banks/src/host.cpp

dj-park, 2023-07-21:
@randyh62 
I am really sorry for the confusion. I retried and the old OpenCL host code from 21.2 worked for both ZCU102 and ZCU102_DFX platform.

closed 2023-07-26 by dj-park