Xilinx/Vitis-Tutorials #407: MMAP issue for vector addition example

RatkoFri, 2023-08-07:
** Which tutorial are you running? **
https://github.com/Xilinx/Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/Part4-data_center.md

** Describe the issue **
After running software emulation for alveo u280 i get following issue: 

```
[user@host$ ./app.exe 
argc = 1
argv[0] = ./app.exe
Open the device0
Load the xclbin ./vadd.xclbin
Kernel Name: vadd_1, CU Number: 0, Thread creation status: success
MMAP_FAILED: Invalid argument
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to mmap file: /xxx/xxx/xxx/xxx/Vitis-Tutorials/Getting_Started/Vitis/example/u250/sw_emu/.run/102438/sw_emu/device0/memory_map_file(Invalid argument)
Software emulation of compute unit(s) exited unexpectedly
Terminated
```

I am using Vitis 2023.1 and appropriate platform files. 

** To Reproduce **

- change directory

```
cd <Path to the cloned repo>/Getting_Started/Vitis/example/u250
mkdir sw_emu
cp xrt.ini sw_emu
cd sw_emu
```

- compile 

```
g++ -g -std=c++17 -Wall -O0 ../../src/host.cpp -o ./app.exe -I$XILINX_XRT/include/ -L$XILINX_XRT/lib -lxrt_coreutil -pthread
emconfigutil --platform xilinx_u280_gen3x16_xdma_1_202211_1
v++ -c -t sw_emu --platform xilinx_u280_gen3x16_xdma_1_202211_1 --config ../../src/u250.cfg -k vadd -I../../src ../../src/vadd.cpp -o ./vadd.xo 
v++ -l -t sw_emu --platform xilinx_u280_gen3x16_xdma_1_202211_1 --config ../../src/u250.cfg ./vadd.xo -o ./vadd.xclbin

```
- run application 

```
export XCL_EMULATION_MODE=sw_emu
./app.exe

```

** Your environment: **

Station and 

```
System Configuration
  OS Name              : Linux
  Release              : 4.18.0-492.el8.x86_64
  Version              : #1 SMP Tue May 9 17:56:55 UTC 2023
  Machine              : x86_64
  CPU Cores            : 256
  Memory               : 483001 MB
  Distribution         : CentOS Stream 8
  GLIBC                : 2.28
  Model                : AS-4124GS-TNR

XRT
  Version              : 2.16.0
  Branch               : master
  Hash                 : 6eff2fc2878a1e35061bfc2c3ef83eee0df7f112
  Hash Date            : 2023-06-27 15:01:22
  XOCL                 : 2.16.0, 6eff2fc2878a1e35061bfc2c3ef83eee0df7f112
  XCLMGMT              : 2.16.0, 6eff2fc2878a1e35061bfc2c3ef83eee0df7f112
```

V++ and Vitis 

```
****** v++ v2023.1 (64-bit)
  **** SW Build 3860322 on 2023-05-04-06:32:48
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2023 Advanced Micro Devices, Inc. All Rights Reserved.

****** Vitis Development Environment
****** Vitis v2023.1 (64-bit)
  **** SW Build 3860322 on 2023-05-04-06:32:46
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2023 Advanced Micro Devices, Inc. All Rights Reserved.
```

randyh62, 2023-08-08:
Do you see this same issue when you use the U250 platform?

RatkoFri, 2023-08-10:
I just tested with the U250 platform and I have the same issue.

RatkoFri, 2023-08-21:
Solved. 

This error arises due to insufficient space on storage. It has nothing to do with Vitis tool.

closed 2023-08-21 by RatkoFri