Xilinx/Vitis-Tutorials #408: can i add new gnu to vitis ide

sususjysjy, 2023-08-12:
i am using vitis ide,but i want use riscv-toolchain gnu,can i add new gnu to vitis ide?

imrickysu, 2023-10-16:
Hi @sususjysjy , may I know your use case of adding riscv-toolchain to Vitis IDE?

Currently we don't provide this feature but if you can share more details maybe we can add this feature in the future.

Since this is not related to the tutorials in this repo directly, I'd like to close this issue in a week but recommend you to share your ideas to https://support.xilinx.com/s/topiccatalog?language=en_US.

sususjysjy, 2023-10-16:
> 你好@sususjysjy，我可以知道您将 riscv-toolchain 添加到 Vitis IDE 的用例吗？
> 
> 目前我们不提供此功能，但如果您可以分享更多详细信息，也许我们将来可以添加此功能。
> 
> 由于这与本存储库中的教程没有直接关系，因此我想在一周内关闭此问题，但建议您将您的想法分享到https://support.xilinx.com/s/topiccatalog?language=en_US。

Thank you for your reply. As I want to port the rv soft core to the xilinx FPGA board, I need the rv toolchain

AnusheelXilinx, 2024-12-21:
Hi @kevinsu20, if there are any open queries, please feel free to post it in Vitis Community Forums:
https://adaptivesupport.amd.com/s/topic/0TO2E000000YKXhWAO/vitis?language=en_US

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx