Xilinx/Vitis-Tutorials #410: Failed Emulation Hardware on a vck190

Dalhfire, 2023-09-04:
Hi, I was trying to build the project following the tutorial and i got an error in the Emulation-HW build process.
The error is the following :

[image]

Do someone knows how to resolve this issue ?
Thanks in advance,
Dalhfire

randyh62, 2023-09-04:
What tutorial are you working through? Does the exa1 file exist at the specified path?

randyh62, 2023-09-12:
It looks like the exa1 is the name of the PS application in your build. I am not sure why it can't find it or why it is looking for it there. Did you ever try to build it for the U250 card as it is written?

Dalhfire, 2023-09-15:
Yes it is and no currently i am working on a Versal vck190 card and I do not know if it works the same

randyh62, 2023-09-15:
So, the exa1 file needs to be specified as part of the --package command to get it on the SD card. --package.sd_file I think. You can add this command to point to where the exa1 file is since it is not where the tool is looking. 
https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Packaging-for-Embedded-Platforms

The host application and the RTL kernel should run on the Versal device.

randyh62, 2023-09-15:
I am closing this issue as you are creating your own path. This is not an issue with the tutorial.

closed 2023-10-16 by imrickysu