Xilinx/Vitis-Tutorials #411: Need help for tutorial 04-Polyphase-Channelizer

SimoneDag99, 2023-09-06:
Hi to everyone. I'm studying the implementation of the channelizer in this tutorial and I cannot understand two things in "polyphase_fir_app.cpp".
First: where "taps_M16_init.h" comes from?
Second: line 14, how does it work without arguments specified?

Thanks you for your replies.

xflorentw, 2023-09-20:
Hi @SimoneDag99 

The file polyphase_fir_app.cpp is not compiled as part of the full system. polyphase_fir_graph is a subgraph and the file polyphase_fir_app.cpp is used to test it independently.
But as part of the full system polyphase_fir_app.cpp is not used, thus even if taps_M16_init.h the full system is compiling

AnusheelXilinx, 2024-12-21:
Query addressed. Closing the issue. 

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx