Xilinx/Vitis-Tutorials #414: Vitis-Tutorials/Developer_Contributed /01-Versal_Custom_Thin_Platform_Extensible_System/: Petalinux Makefile not works as expected on Readme.md: Petalinux Makefile not works as expected on Readme.md

GianlucaGaroia, 2023-10-18:
Running Tutorial:
2023.1/Vitis-Tutorials/Developer_Contributed/01-Versal_Custom_Thin_Platform_Extensible_System/

Issue:
Error at the petalinux build step:
make[2]: ./src/config: Permission denied
make[2]: *** [Makefile:57: create] Error 127

Steps to reproduce:
just followed Readme.md instructions and run "make all" from root folder as first compilation - environment is Ubuntu 22.04.2 with Vivado/Vitis/Petalinux 2023.1 correctly installed (step 1 of the Makefile does version check with no error).
Step 2 and 3 of Makefile also runs correctly, problem is at step 4 that errors out in trying to access this file that is reported as "Permission Denied":
<root folder>/linux/petalinux/src/config 

By making it executable (chmod +x <root folder>/linux/petalinux/src/config) the error changes to:
./src/config: 61: Bad substitution

That makes me think this config file and/or its usage has issues

imrickysu, 2023-10-19:
Could you try `chmod -R 777` for your working directory?

GianlucaGaroia, 2023-10-19:
This gives same result as when adding +x permission to the mentioned ./src/config file:

./src/config: 61: Bad substitution

Thanks in advance

We continuously commit to comply with the applicable data protection laws and ensure fair and transparent processing of your personal data. 
Please read our privacy statement including an information notice and data protection policy for detailed information on our website.

BrunoVandeveldeXilinx, 2023-10-23:
Hi, this is normal since that config file is an executable. If you fork or clone the tutorial then it works fine, it's only when you zip/unzip it that you will have to set the permission right. Let me look if I can fix it in the Makefile-flow itself, so this issue does not pops up again :-)

AnusheelXilinx, 2024-12-20:
Hi @BrunoVandeveldeXilinx - is this fixed in the latest release of the tutorial?

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx