Xilinx/Vitis-Tutorials #415: zcu102 cannot boot from SD card

zhangapeng, 2023-10-21:
When I followed the tutorial ”Vitis-Tutorials/blob/2023.1/Getting_Started/Vitis/Part4-embedded_platform.md“ to operate on the GUI of vitis, the generated sd.img could not be started after burning. I use the base platform when building the project, and then the domain was also set up. There was no problem in compiling after the hardware was built，generate sdcard, but cannot boot from the SD card after burning it, how should I solve it?

sven2314, 2023-12-10:
@zhangapeng Hi, Before trying to boot from SD card, I think you need to set the boot MODE to SD mode.  Besides, do you observe any outputs from UART terminal of zcu102. If so, Could you provide the output log from UART terminal so that we could do further investigation. Thanks.

imrickysu, 2024-01-26:
Hi @zhangapeng , this topic requires more details from your side. 

Since we didn't hear back from you, I'm closing this thread for now. Feel free to reopen it if you can add more details.

closed 2024-01-26 by imrickysu