Xilinx/Vitis-Tutorials #416: 10-GeMM_AIEvsDSP/AIE failed to meet timing

ChengyueWang, 2023-10-29:
For the example: Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/10-GeMM_AIEvsDSP/AIE
When I run:
make run TARGET=hw ITER_CNT=16 GEMM_SIZE=32
The error occurs saying that the deign failed to meet timing. I am using the Vitis 2023.1 and on ubuntu 2020.
Which compile flag makes it easiest for the design to meet timing? 

ERROR: [VPL 101-2] design did not meet timing - Design failed to meet timing.
    Failed timing checks (paths):
        {vitis_design_i/ai_engine_0/inst/pl_ai_ch_19/inst/pl2ai_wrapper_128/bli0.bli_s0o_tready_1_reg/C --> vitis_design_i/ai_engine_0/inst/pl_ai_ch_19/inst/pl2ai_wrapper_128/mrs_s0/FSM_sequential_state_reg[0]_fret_replica/CE}

    Please check the routed checkpoint (dr_routed_timing.dcp) and timing summary report (dr_timing_summary.rpt) for more information.
ERROR: [VPL 101-3] sourcing script /home/chengyue/github/Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/10-GeMM_AIEvsDSP/AIE/build/gemm_32x32x32/x1/hw/_x/link/vivado/vpl/scripts/impl_1/_full_write_device_image_pre.tcl failed
INFO: [VPL 17-206] Exiting Vivado at Sun Oct 29 15:32:04 2023...
[Sun Oct 29 15:32:09 2023] impl_1 finished
ERROR: [VPL 12-13638] Failed runs(s) : 'impl_1'
[15:32:09] Run vpl: Step impl: Failed
INFO: [VPL 17-206] Exiting Vivado at Sun Oct 29 15:32:09 2023...
[15:32:09] Run vpl: FINISHED. Run Status: impl ERROR

===>The following messages were generated while  Compiling (bitstream) accelerator binary: vck190_aie_gemm.hw Log file: /home/chengyue/github/Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/10-GeMM_AIEvsDSP/AIE/build/gemm_32x32x32/x1/hw/_x/link/vivado/vpl/prj/prj.runs/impl_1/runme.log  :
ERROR: [VPL 101-2] design did not meet timing - Design failed to meet timing.
    Failed timing checks (paths):
        {vitis_design_i/ai_engine_0/inst/pl_ai_ch_19/inst/pl2ai_wrapper_128/bli0.bli_s0o_tready_1_reg/C --> vitis_design_i/ai_engine_0/inst/pl_ai_ch_19/inst/pl2ai_wrapper_128/mrs_s0/FSM_sequential_state_reg[0]_fret_replica/CE}

    Please check the routed checkpoint (dr_routed_timing.dcp) and timing summary report (dr_timing_summary.rpt) for more information.
ERROR: [VPL 101-3] sourcing script /home/chengyue/github/Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/10-GeMM_AIEvsDSP/AIE/build/gemm_32x32x32/x1/hw/_x/link/vivado/vpl/scripts/impl_1/_full_write_device_image_pre.tcl failed

===>The following messages were generated while  creating FPGA bitstream. Log file: /home/chengyue/github/Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/10-GeMM_AIEvsDSP/AIE/build/gemm_32x32x32/x1/hw/_x/link/vivado/vpl/runme.log :
ERROR: [VPL 12-13638] Failed runs(s) : 'impl_1'
ERROR: [VPL 60-773] In '/home/chengyue/github/Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/10-GeMM_AIEvsDSP/AIE/build/gemm_32x32x32/x1/hw/_x/link/vivado/vpl/vivado.log', caught Tcl error:  ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, Failed to complete hardware generation. The run name is 'impl_1'. An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [15:32:10] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:15 ; elapsed = 00:40:48 . Memory (MB): peak = 448.195 ; gain = 0.000 ; free physical = 33015 ; free virtual = 50242
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.

Chenyu617236, 2023-11-09:
i met the same problem...

allyzhou, 2023-12-05:
Which Vitis version are you using?

AnusheelXilinx, 2024-12-20:
Hi @ChengyueWang, please try the latest release of the tutorial and let us know if you still see an issue. 

Thanks
Anusheel

closed 2024-12-20 by AnusheelXilinx