Xilinx/Vitis-Tutorials #419: Vitis-Tutorials/AI_Engine_Development/AIE-ML/Design_Tutorials/01-AIE-ML-programming-and-optimization

strubel001, 2023-11-23:
The command "make OPT=0 all_hw_emu" as given in the tutorial doesn't work.
Probably the command "build_hw_emu" sould be used instead.

Joel-De, 2024-01-11:
@strubel001 Were you able to get hw_emu working for this tutorial, I'm still running into weird issues and was wondering if it was just me?

OTremois, 2024-01-30:
I just Changed the README file:

- build_hw_emu builds and packages the files for hw emulation
- launch_hw_emu... lauches the hw emulation
- run_hw_emu : builds and launch the hw emulation

During hw emulation a lot of lines similar to : **[  354.714956] spi-nor spi0.0: Unknown command extension type**
are displayed.
The simulation is launched afterwards and suvvesful.

closed 2024-08-14 by allyzhou