Xilinx/Vitis-Tutorials #426: Issue with Bit Width Exceeding Limit in HLS Stream as Intermediate Variable | Vitis 2023.2

hanghug, 2023-12-28:
Hello,

I am encountering an issue while using **hls::stream** as an intermediate variable in my project. **The compiler is indicating that the bit width is too large, exceeding the 4096 limit.** Here's the context of my problem:

I created my own data structure, similar to hls::vector. Below is the relevant code snippet:
```cpp
#ifdef __SYNTHESIS__
#define SYN_PRAGMA(PRAG) _Pragma(#PRAG)
#else
#define SYN_PRAGMA(PRAG)
#endif

template <typename T, unsigned N>
struct array {
    typedef T value_type;
    static const unsigned size = N;

    T data[N];

protected:
  INLINE void pragma() const {
    SYN_PRAGMA(HLS DISAGGREGATE variable=this)
  }

public:
    INLINE array() {
       pragma();
       SYN_PRAGMA(HLS ARRAY_PARTITION variable=this->data complete)
    }

    // ... [Additional member functions]
};
```
Then, I declared a data type as follows:
```cpp
typedef array<ap_fixed<32,8>, 512> input_t;
hls::stream<input_t> data_recv;
#pragma HLS STREAM variable=data_recv
```
 However, when I declare a data stream wrapped with the input_t type inside a function, I encounter the aforementioned error, From the log file, I noticed the message: 
"**INFO: [HLS 214-241] Aggregating fifo (hls::stream) variable 'data_recv' with compact=bit mode**."It seems to perform an Aggregate operation.
 Is there a way to perform Disaggregate? I have not encountered this error when using Vivado HLS previously.

imrickysu, 2024-01-08:
Hi @hanghug , is this issue related to a specific tutorial? If so, could you point out the tutorial link? If not, could you please direct to forums.xilinx.com for tech support? Moderators there will be more experienced to help you with this specific question. Thanks.

hanghug, 2024-01-08:
Thank you for your responses.I have already directed my query to forums.xilinx.com for tech support, but I haven't yet received a satisfactory solution.
&nbsp;

陆亿行
***@***.***

&nbsp;

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Ricky ***@***.***&gt;; 
发送时间: 2024年1月8日(星期一) 上午10:47
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [Xilinx/Vitis-Tutorials]  Issue with Bit Width Exceeding Limit in HLS Stream as Intermediate Variable | Vitis 2023.2 (Issue #426)

***@***.***，这个问题与某个具体的教程有关吗？如果是的话，你能指出教程链接吗？如果没有，请直接到forums.xilinx.com获取技术支持？版主那里会有更有经验的人来帮助你解决这个具体的问题。谢了。

-
直接回复这封邮件，在GitHub上查看，或取消订阅.
你收到这个是因为你被提到了。消息ID:<赛灵克斯/葡萄酒条约/问题/ 426 / ***@***.***和&gt;

AnusheelXilinx, 2024-12-20:
Closing as there are no further open questions in the thread. 

thanks
Anusheel

closed 2024-12-20 by AnusheelXilinx