Xilinx/Vitis-Tutorials #440: Dead lock in tutorial "01-rtl_kernel_workflow"

peter997799, 2024-05-29:
Path to the Tutorial: Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow
     Hi everyone, I am new to Vitis RTL Kernel and want to develop my project base on this tutorial. However when I finish compiling the provided codes and run the kernel with instruction `make run TARGET=hw_emu HOST=user` the kernel fall into an infinite loop and won't give the result. On my terminal the host code does still shows `INFO: IP Start` which means the host program has at least sucessfully executed to line 87 of user-host.cpp, but the condition that required to jump out the while loop from line 92 to 98 `while (krnl_done != IP_DONE) {` is never met. Which means IP_DONE signal is never be set. Is there any things I did wrong? 

[image]

During the compilation, there are some warning messages printed out on the terminal which I think could cause the problem:
`WARNING: [Vivado 12-12407] VLNV in kernel.xml does not match VLNV in any of the IPs specified with the ip_directory option: mycompany.com:kernel:Vadd_A_B:1.0
`

There are also 34 warnings while checking VPL:
`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP pfm_top_axi_vip_0_0 is locked. The IP is locked for the following reason - * IP definition 'AXI Verification IP (1.1)' for IP 'pfm_top_axi_vip_0_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`  

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP pfm_top_connect_to_es_cu_0 is locked. The IP is locked for the following reason - * IP definition 'AXI Interconnect (2.1)' for IP 'pfm_top_connect_to_es_cu_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP pfm_top_smartconnect_0_0 is locked. The IP is locked for the following reason - * IP definition 'AXI SmartConnect (1.0)' for IP 'pfm_top_smartconnect_0_0' (customized with software release 2022.1) has a different revision in the IP Catalog. * IP 'pfm_top_smartconnect_0_0' contains one or more locked subcores.* Target IP definition 'AXI SmartConnect (1.0)' requires a revision change. Please review the change log before upgrading the IP.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP pfm_top_xbar_0 is locked. The IP is locked for the following reason - * IP definition 'AXI Crossbar (2.1)' for IP 'pfm_top_xbar_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_arsw_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Switchboard (1.0)' for IP 'bd_387c_arsw_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_awsw_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Switchboard (1.0)' for IP 'bd_387c_awsw_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_bsw_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Switchboard (1.0)' for IP 'bd_387c_bsw_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00arn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_m00arn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00awn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_m00awn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00bn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_m00bn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00e_0 is locked. The IP is locked for the following reason - * IP definition 'SC EXIT (1.0)' for IP 'bd_387c_m00e_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00rn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_m00rn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00s2a_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect SC2AXI Bridge (1.0)' for IP 'bd_387c_m00s2a_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m00wn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_m00wn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_m01arn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_m01arn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

[... 59 lines cut ...]

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_sbn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_sbn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_srn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_srn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_swn_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Node (1.0)' for IP 'bd_387c_swn_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

`
WARNING: [IP-LOCK-01] In your vivado project:prj for design:dr the IP bd_387c_wsw_0 is locked. The IP is locked for the following reason - * IP definition 'SmartConnect Switchboard (1.0)' for IP 'bd_387c_wsw_0' (customized with software release 2022.1) has a different revision in the IP Catalog.
`

Here are my work environment OS/program/device:

OS: Ubuntu 20.04.6 LTS
Vitis version: 2023.2 
XRT Version: 2.16.204
device: xilinx_u55c_gen3x16_xdma_3_202210_1

Any help would be greatly appreciated - please let me know if more info is needed!

siujeong, 2024-10-17:
Did you solve this issue,,?  I'm having the same issue now,,

AnusheelXilinx, 2025-01-10:
This tutorial was tested with xilinx_u250_gen3x16_xdma_4_1_202210_1 platform. 

Thanks
Anusheel