Xilinx/Vitis-Tutorials #441: Vitis_HLS/Feature_Tutorials/02-Beamformer_Analysis

qingqins, 2024-06-03:
Csim failed with below error.
Add '`#include <cstdlib>`' in file [beamformer_tb.cpp](https://github.com/Xilinx/Vitis-Tutorials/blob/2023.2/Vitis_HLS/Feature_Tutorials/02-Beamformer_Analysis/reference_files/beamformer_tb.cpp) can fix the issue.

INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
INFO: [HLS 200-2036] Building debug C Simulation binaries
   Compiling ../../../../reference_files/beamformer_tb.cpp in debug mode
../../../../reference_files/beamformer_tb.cpp:5056:11: error: use of undeclared identifier 'system'
        retval = system("diff --brief -w result.dat result.golden_float.dat");
                 ^
1 error generated.

allyzhou, 2024-08-14:
@qingqins I've tried on both Vitis Unified IDE 2023.2 and 2024.1. without adding '#include <cstdlib>' in file beamformer_tb.cpp , C-Sim can complete successfully without any error. 

> 

  -225.000000 1865.000000 
  -300.000000 1970.000000 
  beamso_i   beamso_q
  -75.000000 1745.000000 
  -150.000000 1790.000000 
  -225.000000 1865.000000 
 Test passed !
 INFO: [SIM 211-1] CSim done with 0 errors.
 INFO: [SIM 211-3] *************** CSIM finish ***************
 INFO: [HLS 200-111] Finished Command csim_design CPU user time: 0.4 seconds. CPU system time: 0.35 seconds. Elapsed time: 0.82 seconds; current allocated memory: 0.000 MB.
 INFO: [HLS 200-1510] Running: close_project 
 INFO: [HLS 200-112] Total CPU user time: 41.43 seconds. Total CPU system time: 1.17 seconds. Total elapsed time: 44.28 seconds; peak allocated memory: 483.523 MB.
 INFO: [Common 17-206] Exiting vitis_hls at Wed Aug 14 00:49:07 2024...
 INFO: [vitis-run 60-791] Total elapsed time: 0h 0m 49s
 C-simulation finished successfully

qingqins, 2024-08-14:
Hi @allyzhou ,
Thanks for checking.
The issue is fixed in latest code. I close this issue.
Previous only include one file:
`#include "beamformer.h"`

Latest:
`#include <stdio.h>`
`#include <math.h>`
`#include "beamformer.h"`