Xilinx/Vitis-Tutorials #444: Hardware_Acceleration/Design_Tutorials/06-cholesky-accel

htprofessor, 2024-06-17:
Excuse me
this cholesky algorithm can solve the matrix of 512,what should I do to solve larger matrix like 1024 or more.
I changed MAXN and NCU ,but it doesnt work.
[image]
[image]

allyzhou, 2024-07-02:
Hi @AnusheelXilinx  can you take a look at this issue. thanks

AnusheelXilinx, 2024-07-04:
@htprofessor, If I run the "make TARGET=sw_emu run" after changing the "MAXN" and "NCU", I do not see any issues.

Thanks
Anusheel

AnusheelXilinx, 2024-12-20:
Closing the issue as there are no further concerns. 

Thanks
Anusheel