Xilinx/Vitis-Tutorials #446: Issue in AI_Engine_Development/AIE/Feature_Tutorials/16-external-traffic-generator-aie/SW-HW_Emulation

qingqins, 2024-06-26:
There are two 'syn.cflags' defined in below two cfg files.
HLS synthesis will fail with error "ERROR: [HLS 200-1923] Found duplicate ini entry [hls] syn.cflags"

https://github.com/Xilinx/Vitis-Tutorials/blob/2024.1/AI_Engine_Development/AIE/Feature_Tutorials/16-external-traffic-generator-aie/SW-HW_Emulation/pl_kernels/s2mm.cfg

https://github.com/Xilinx/Vitis-Tutorials/blob/2024.1/AI_Engine_Development/AIE/Feature_Tutorials/16-external-traffic-generator-aie/SW-HW_Emulation/pl_kernels/mm2s.cfg

allyzhou, 2024-06-27:
Thanks for reporting the issue, the owner has been notified and currently working on that.

allyzhou, 2024-07-12:
The error has been fixed via the commit 679390bfa60503b009c048d5e618445c55ad8f84

closed 2024-07-12 by allyzhou