Xilinx/Vitis-Tutorials #447: Vitis_Platform_Creation 01-Edge-KV260

yakyol, 2024-06-26:
https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/Vitis_Platform_Creation/Design_Tutorials)/[01-Edge-KV260]/step2.md

xsct
createdts -hw ../kv260_hardware_platform/kv260_hardware_platform.xsa -zocl -out . -platform-name mydevice -git-branch xlnx_rel_v2022.1 -overlay -compile

Getting the results;
zynqmp.dtsi:439.38-449.5: Warning (interrupt_provider): /axi/interrupt-controller@f9010000: Missing #address-cells in interrupt provider
  also defined at pcw.dtsi:11.6-14.3
zynqmp.dtsi:649.23-660.5: Warning (interrupt_provider): /axi/gpio@ff0a0000: Missing #address-cells in interrupt provider
  also defined at zynqmp-clk-ccf.dtsi:200.7-202.3
  also defined at pcw.dtsi:83.7-88.3
INFO: Device tree generation successful

than next step
dtc -@ -O dtb -o mydevice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtbo mydevice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtsi

fails;
mydevice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtsi:39.44-51.4: Warning (interrupt_provider): /fragment@1/__overlay__/interrupt-controller@80000000: Missing #address-cells in interrupt provider
please advice

allyzhou, 2024-06-27:
Thanks for reporting the issue, the owner has been notified and currently working on that.

sven2314, 2024-07-08:
@yakyol I cannot reproduce your issue. Could you provide your XSA file for further investigation or change to latest version to see whether issue exists or not.

yakyol, 2024-07-08:
I use the steps of https://github.com/Xilinx/Vitis-Tutorials/tree/2023.1/Vitis_Platform_Creation

my OS environment
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic

Versions;
Vivado v2023.1.2 (64-bit) SW Build: 3954437 on Wed Aug 9 23:05:19 MDT 2023 IP Build: 3954312 on Thu Aug 10 04:10:56 MDT 2023 SharedData Build: 3952698 on Tue Aug 08 21:36:00 MDT 2023

Vitis 2023.1

[kv_260_harware_xsa.tar.gz](https://github.com/user-attachments/files/16126993/kv_260_harware_xsa.tar.gz)

sven2314, 2024-08-14:
@yakyol 
Hi, I can create DTS and DTBO based on your XSA file. Please see following log.
`zynqmp.dtsi:439.38-449.5: Warning (interrupt_provider): /axi/interrupt-controller@f9010000: Missing #address-cells in interrupt provider
  also defined at pcw.dtsi:11.6-14.3
zynqmp.dtsi:649.23-660.5: Warning (interrupt_provider): /axi/gpio@ff0a0000: Missing #address-cells in interrupt provider
  also defined at zynqmp-clk-ccf.dtsi:200.7-202.3
  also defined at pcw.dtsi:83.7-88.3
INFO: Device tree generation successful
 make dtbo
dtc -@ -O dtb -o github/Vitis-Tutorials/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/ref_files/step2_pfm/mydeivice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtbo github/Vitis-Tutorials/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/ref_files/step2_pfm/mydeivice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtsi
github/Vitis-Tutorials/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/ref_files/step2_pfm/mydeivice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtsi:39.44-51.4: Warning (interrupt_provider): /fragment@1/__overlay__/interrupt-controller@85000000: Missing #address-cells in interrupt provider
mkdir -p dtbo_output && cp github/Vitis-Tutorials/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/ref_files/step2_pfm/mydeivice/psu_cortexa53_0/device_tree_domain/bsp/pl.dtbo dtbo_output/`

Following is my ENV：
`LSB Version:    core-11.1.0ubuntu4-noarch:printing-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy`
I think maybe there is something wrong with your ENV.
Please clean your workspace and try again.

AnusheelXilinx, 2024-12-20:
Closing the issue as there are no further concerns. 

Thanks
Anusheel

closed 2024-12-20 by AnusheelXilinx