Xilinx/Vitis-Tutorials #449: Missing image

xilinx-nkrueger, 2024-07-12:
https://github.com/Xilinx/Vitis-Tutorials/blob/58e54fa809d6cdb4a59212055aa1d23e051f7ec7/Getting_Started/Vitis/Part4-embedded_platform.md?plain=1#L18

The image link from this tutorial is broken.

imrickysu, 2024-08-02:
This is a public image in UG1393
Image: https://docs.amd.com/viewer/attachment/2AHmluw_4m~0LHemYekb6A/KyHSdLDPWS_frtia6T0YuA
Parent chapter: https://docs.amd.com/r/en-US/ug1393-vitis-application-acceleration/Introduction-to-the-Vitis-Unified-IDE

@xilinx-nkrueger Thanks for reporting this issue.
@sven2314  Please update the image link.

xilinx-nkrueger, 2024-08-02:
Interesting... If paste the image URL into the browser the image loads up, but if I go to the readme, I see this:

[image]

And if I click the link, I see this:

[image]

I'm on the AMD VPN.

AnusheelXilinx, 2024-12-20:
I can see the correct image with URL corrected to github.com. Closing the issue.

closed 2024-12-20 by AnusheelXilinx