Xilinx/Vitis-Tutorials #461: xilinx_u200_gen3x16_xdma_1_202110_1 not available

Sai-222777, 2024-10-17:
I am following Hardware Acceleration-Design Tutorials, and in the make_options.mk, there is a requirement for 
PLATFORM ?= xilinx_u200_gen3x16_xdma_1_202110_1,.
In the AMD website there is only xilinx_u200_gen3x16_xdma_2_202110_1 available?
From where do I download or obtain the xilinx_u200_gen3x16_xdma_1_20210_1?

AnusheelXilinx, 2024-12-20:
Hi @Sai-222777, please use the below link to download the latest platform xilinx_u200_gen3x16_xdma_2_202110_1:
https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/alveo/u200.html

Are you seeing any errors while building the platform with xilinx_u200_gen3x16_xdma_2_202110_1?

Thanks
Anusheel

Sai-222777, 2024-12-21:
I have already tried, and am getting errors.
I need specifically the older version.

Thank you,

https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/alveo/u200.html

AnusheelXilinx, 2024-12-23:
Hi @Sai-222777, 
I have tested 01-Convolution-Tutorial with the xilinx_u200_gen3x16_xdma_2_202110_1 and did not see any issue. 
Please let us know the tutorial that you are using and the error/issue while using xilinx_u200_gen3x16_xdma_2_202110_1. 

Thanks
Anusheel

AnusheelXilinx, 2025-01-06:
Hi @Sai-222777, did you get chance to try the 01-Convolution-Tutorial with the xilinx_u200_gen3x16_xdma_2_202110_1 example? 

Thanks
Anusheel

Sai-222777, 2025-01-06:
Hi,
I have tried it, and it doesn't work.. Did it work out for you?

Thanks,
Saishree

AnusheelXilinx, 2025-01-06:
Yes, it works for me @Sai-222777. 
May I know which tutorial did you use, changes made in the tutorial and the error?

Sai-222777, 2025-01-06:
I have been following this tutorial.
Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/01-convolution-tutorial at 2024.1 · Xilinx/Vitis-Tutorials · GitHub<https://github.com/Xilinx/Vitis-Tutorials/tree/2024.1/Hardware_Acceleration/Design_Tutorials/01-convolution-tutorial>

I get errors when I try to "make build run" in hardware mode. I will share the exact error soon. Thank you so much for responding.

closed 2025-01-13 by AnusheelXilinx