Xilinx/Vitis-Tutorials #462: Hardware Acceleration/01-convolution/ make run error -> Filter2DKernel does not work on this part

srisurya-m, 2024-10-17:
I am following the Hardware Acceleration tutorial and in the first task (01 convolution) , i had to download  xilinx_u200_gen3x16_xdma_1_202110_1 package, but unable to find the exact package and hence downloaded,
xilinx_u200_gen3x16_xdma_2_202110_1 package, but getting a error, 
This code was setup/meant for  xilinx_u200_gen3x16_xdma_1_202110_1, please guide on what changes I have to make for it to be compatible for  xilinx_u200_gen3x16_xdma_2_202110_1. Like (part no in .tcl script,) or any other changes. 

Thank you

AnusheelXilinx, 2024-12-20:
Hi @srisurya-m , can you please the full error that you are getting while building the design?
Also, did you try to modify the make_options.mk file with the latest platform? 

Thanks
Anusheel

AnusheelXilinx, 2024-12-22:
Hi @srisurya-m, I just tested the tutorial build run with xilinx_u200_gen3x16_xdma_2_202110_1 and did not see any issue. 
Please update the make_options.mk with latest platform:
PLATFORM ?= xilinx_u200_gen3x16_xdma_2_202110_1

Thanks
Anusheel

closed 2024-12-22 by AnusheelXilinx