Xilinx/Vitis-Tutorials #463: VitisFast Track

GuoZhongYi-Student, 2024-12-01:
```
Compiling XilPLMI Library
/tools/Xilinx/Vivado/2022.2/gnu/microblaze/lin/bin/../x86_64-oesdk-linux/usr/bin/microblaze-xilinx-elf/microblaze-xilinx-elf-gcc.real: error while loading shared libraries: libwebp.so.6: cannot open shared object file: No such file or directory
make[3]: *** [Makefile:50: xplmi_err_common.o] Error 127
make[2]: *** [Makefile:46: CIPS_0_pspmc_0_psv_pmc_0/libsrc/xilplmi_v1_7/src/make.libs] Error 2
make[1]: *** [Makefile:18: all] Error 2
ERROR: [#UNDEF] Running Make Failed
ERROR: [Hsi 55-1444] Error(s) while running make
INFO: [Project 1-1945] The Hardware Platform can be used for Hardware and Hardware Emulation
INFO: [Project 1-1941] Successfully created Hardware Platform: /home/gzy/Vitis-Tutorials/Getting_Started/Vitis_Platform/ref_files/build/vivado/vck190_custom_hw.xsa
write_hw_platform: Time (s): cpu = 00:00:12 ; elapsed = 00:00:12 . Memory (MB): peak = 4650.809 ; gain = 992.988 ; free physical = 37995 ; free virtual = 56008
ERROR: [Common 17-39] 'write_hw_platform' failed due to earlier errors.
```

AnusheelXilinx, 2024-12-20:
Hi @sven2314, can you please look into this and share your thoughts?

Thanks
Anusheel

sven2314, 2025-01-06:
@GuoZhongYi-Student 
Please refer to the blog, which contains a solution for your issue: https://adaptivesupport.amd.com/s/question/0D54U00008sdvpcSAA/vitis20222-getting-started-fast-track-error-libwebpso6?language=en_US

closed 2025-01-06 by AnusheelXilinx