Xilinx/Vitis-Tutorials #465: 10-GeMM_AIEvsDSP failed to compile after running the build command. #Vitis2023.2

Prathmesh-shalu141, 2024-12-20:
The issue I am facing is given below.

/home/cselab/Downloads/Vitis-Tutorials-2023.2/AI_Engine_Development/AIE/Design_Tutorials/10-GeMM_AIEvsDSP/AIE/design/aie_src/graph.h:11:13: fatal error: matrix_mult_graph.hpp: No such file or directory
    #include "matrix_mult_graph.hpp"
             ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

AnusheelXilinx, 2024-12-20:
Hi @Prathmesh-shalu141, did you set the DSPLIB_VITIS correctly? 
As mentioned here:https://github.com/Xilinx/Vitis-Tutorials/tree/2023.2/AI_Engine_Development/AIE/Design_Tutorials/10-GeMM_AIEvsDSP#Before-you-Begin:~:text=sdk.sh%20script.-,DSPLIB_VITIS,-is%20the%20path

Thanks
Anusheel

Prathmesh-shalu141, 2024-12-21:
Hi @AnusheelXilinx  the sample environment setup file is as follows :

# =======================================================
# Set Platform ,Vitis and Versal Image repo
# =======================================================
export PLATFORM_REPO_PATHS=/home/cselab/IMP_Files/tools/Vitis/2023.2/base_platforms
export XILINX_VITIS=/home/cselab/IMP_Files/tools/Vitis/2023.2
export COMMON_IMAGE_VERSAL=/opt/petalinux/2023.2

# ====================================================
# Source Versal Image ,Vitis and Aietools
# ====================================================
# Run the below command to setup environment and CXX
source $COMMON_IMAGE_VERSAL/environment-setup-cortexa72-cortexa53-xilinx-linux
source $XILINX_VITIS/settings64.sh
# ========================================================
# Set DSP Library for Vitis
# ========================================================
export DSPLIB_VITIS=/home/cselab/IMP_Files/tools/Vitis/2023.2
# =========================================================
# Platform Selection...
# =========================================================
tgt_plat=xilinx_vck190_base_202320_1
export PLATFORM=$PLATFORM_REPO_PATHS/$tgt_plat/$tgt_plat\.xpfm
# ==========================================================
# Validating Tool Installation
# ==========================================================
echo ""
echo "Aiecompiler:"
which aiecompiler
echo ""
echo "Vivado:"
which vivado
echo ""
echo "Vitis:"
which vitis
echo ""
echo "Vitis HLS:"
which vitis_hls
echo ""
echo ""
echo "DSPLIBS"
echo "$DSPLIB_VITIS"
echo ""

The Vitis 2023.2 path is attached as screen shot below : 

[image]

AnusheelXilinx, 2024-12-21:
@Prathmesh-shalu141, the below ENV is incorrect:
export DSPLIB_VITIS=/home/cselab/IMP_Files/tools/Vitis/2023.2

Please git clone the correct Vitis_Libraries from: https://github.com/Xilinx/Vitis_Libraries/
export DSPLIB_VITIS=<Your Machine Path>/Vitis_Libraries

Thanks
Anusheel

Prathmesh-shalu141, 2024-12-21:
Hi @AnusheelXilinx, I have added the correct vitis libraries, and its working now, but then while creating a hardware image, it failed again. 
I have attached a screen shot of the error I got after running the make command. Please look into it.
[image]

AnusheelXilinx, 2024-12-21:
Hi @Prathmesh-shalu141, the COMMON_VERSAL_IMAGE ENV is incorrect. 
Please refer to below example as reference:
https://github.com/Xilinx/Vitis-Tutorials/blob/bc0458ea062b3372d1e4b01b84b475f434f8b1d8/AI_Engine_Development/AIE/Design_Tutorials/03-beamforming/sample_env_setup.sh#L14

Thanks
Anusheel

Prathmesh-shalu141, 2024-12-21:
Hi @AnusheelXilinx, its working now, Thanks.

AnusheelXilinx, 2024-12-22:
Hi @Prathmesh-shalu141, thank you for the confirmation.