Xilinx/Vitis-Tutorials #469: KV260 Edge Vitis platform tutorial no longer describes how to generate the dtbo file for the platform/application

tom7980, 2025-04-02:
With the update to v2024.2 versions of the tutorials the Vitis Platform KV260 tutorial no longer describes how to correctly generate the dtbo file required to run the application on the starter kit. 

I believe this is an error as it is required to be able to run the applications and is referenced in step 3 as being generated in step 2. Could this be added back in or is there an update to the process to do this?

Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/step2.md

I mostly bring this up as I can't get this example to run on a KR260 running ubuntu which is currently the recommended initial install target for the KR260 platform. Is this example a petalinux only example? If so is there any way we can get an updated version of this for Ubuntu + Vitis + KR260?

tom7980, 2025-04-03:
As an addition to this - the current makefile for this example also doesn't work. The dtbo file generated references an "fpga" symbol in the main device tree but that doesn't exist in the Petalinux or Ubuntu system device tree.

[image]

The platforminfo makefile is also wrong as it is missing the ws folder in the PLATFORM variable inside of it.

https://github.com/Xilinx/Vitis-Tutorials/blob/ca5ceb977627fee2f0d157654cd355cce339cea8/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/ref_files/step3_validate/platforminfo/Makefile#L8

I believe it should be `
PLATFORM=$(ROOT_DIR)/../../step2_pfm/ws/$(PLATFORM_NAME)/export/$(PLATFORM_NAME)/$(PLATFORM_NAME).xpfm 
`

I'd really love to be able to use these boards via the vitis design flow but I just can't seem to figure out what is going wrong or how to generate the correct files to do so (unrelated to these issues my board just hangs and stops responding when I do finally get xmutil to program the xclbin file generated from this tutorial).

tom7980, 2025-04-11:
The issue with the board hanging is related to the generated dtbo file and the axi_intc block in some way - I believe it is causing the kernel to load a driver that is crashing as if I take the interrupt controller out of the dtbo file and load the platform through xmutil I'm able to program the bitstream fine and load the DT overlay.

Claxl, 2025-06-06:
I'm facing the same issue as you. I haven't managed to fix it yet, but it seems like you did. If possible, could you please explain your hotfix in more detail?"

tom7980, 2025-06-06:
Sorry to say that I didn't end up fixing it - even generating the DTBO the old way causes the Kernel to lock up when loading one of the drivers for one of the peripherals. I think it's probably the interrupt controller but I haven't been able to fix that either because the most recent version of petalinux is not supported in yocto yet. 

I would recommend just falling back to an older version of the tools for now as it feels that they are in a transitional period with v2024 that brought with it a lot of changes but also lots of new problems.

Claxl, 2025-06-06:
Can I ask you which version is working for you?

MayerUser, 2025-06-11:
@Claxl  The Version 2023.1 work for me. I also try the version 2024.2, but cannot loadapp. I do not understand why the simple demo vadd did not fully test before issue :(